turtlebot/turtlebot_simulator #46: turtlebot_stdr bringup dies 8 out of 10

jihoonl, 2015-03-19:
```
jihoonl@whoola:/opt/ros/indigo/share/turtlebot_stdr/launch$ roslaunch turtlebot_stdr turtlebot_in_stdr.launch --screen
... logging to /home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/roslaunch-whoola-20955.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://whoola:56301/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: robot0
 * /amcl/global_frame_id: world
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 2.0
 * /amcl/initial_pose_y: 2.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.2
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: map
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: True
 * /cmd_vel_mux/yaml_cfg_file: /opt/ros/indigo/s...
 * /diagnostic_aggregator/analyzers/input_ports/contains: ['Digital Input',...
 * /diagnostic_aggregator/analyzers/input_ports/path: Input Ports
 * /diagnostic_aggregator/analyzers/input_ports/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/input_ports/timeout: 5.0
 * /diagnostic_aggregator/analyzers/input_ports/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/kobuki/contains: ['Watchdog', 'Mot...
 * /diagnostic_aggregator/analyzers/kobuki/path: Kobuki
 * /diagnostic_aggregator/analyzers/kobuki/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/kobuki/timeout: 5.0
 * /diagnostic_aggregator/analyzers/kobuki/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/power/contains: ['Battery', 'Lapt...
 * /diagnostic_aggregator/analyzers/power/path: Power System
 * /diagnostic_aggregator/analyzers/power/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/power/timeout: 5.0
 * /diagnostic_aggregator/analyzers/power/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/sensors/contains: ['Cliff Sensor', ...
 * /diagnostic_aggregator/analyzers/sensors/path: Sensors
 * /diagnostic_aggregator/analyzers/sensors/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
[... 225 lines cut ...]
[robot_manager-2] process has died [pid 21000, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=robot_manager __log:=/home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/robot_manager-2.log].
log file: /home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/robot_manager-2*.log
[robot_spawn_whoola_20955_5924150219310216960-4] process has finished cleanly
log file: /home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/robot_spawn_whoola_20955_5924150219310216960-4*.log
[ INFO] [1426730449.369945113]: Initialising nodelet... [kobuki_safety_controller]
[stdr_gui_node_whoola_20955_4187621309782193888-5] process has died [pid 21161, exit code -11, cmd /opt/ros/indigo/lib/stdr_gui/stdr_gui_node __name:=stdr_gui_node_whoola_20955_4187621309782193888 __log:=/home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/stdr_gui_node_whoola_20955_4187621309782193888-5.log].
log file: /home/jihoonl/.ros/log/c1f702fa-cddb-11e4-9ff5-3085a9948f5e/stdr_gui_node_whoola_20955_4187621309782193888-5*.log
[ INFO] [1426730449.471562484]: Subscribed to map topic.
[ INFO] [1426730449.506439610]: Kobuki initialised. Spinning up update thread ... [kobuki_safety_controller]
[ INFO] [1426730449.506514009]: Nodelet initialised. [kobuki_safety_controller]
[ INFO] [1426730449.667326239]: Received a 775 X 746 map @ 0.020 m/pix

[ INFO] [1426730449.687929227]: Initializing likelihood field model; this can take some time on large maps...
[ INFO] [1426730449.819378923]: Done initializing likelihood field model.
[ INFO] [1426730450.066249515]: Stereo is NOT SUPPORTED
[ INFO] [1426730450.066360690]: OpenGl version: 4.4 (GLSL 4.4).
[ WARN] [1426730454.266330679]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730459.297587092]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730464.328308834]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730464.667563840]: No laser scan received (and thus no pose updates have been published) for 1426730464.667492 seconds.  Verify that data is being published on the /robot0/laser_0 topic.
 [ WARN] [1426730469.359750142]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730474.391097193]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730479.421713928]: Waiting on transform from robot0 to world to become available before running costmap, tf error: 
[ WARN] [1426730479.667116868]: No laser scan received (and thus no pose updates have been published) for 1426730479.667073 seconds.  Verify that data is being published on the /robot0/laser_0 topic.
```

jihoonl, 2015-03-19:
`robot_manager` crashes almost 9 out of 10 times with deb version. Somehow source version fails less.

stonier, 2015-03-20:
I think perhaps we have to drop this. While it looked good, it took alot of work to get it in...and it's still  changing underneath us. We don't have the resources to keep up with it.

jihoonl, 2015-03-20:
Do you mean dropping from documentation only or dropping the deb release as well?

libing64, 2016-04-06:
Is the bug fixed?
The stdr_gui_node dies every time on my computer.  I am using ubuntu14.04 and ros indigo.

[stdr_gui_node_libing0_18101_4696702751935649166-5] process has died [pid 18339, exit code -11, cmd /opt/ros/indigo/lib/stdr_gui/stdr_gui_node __name:=stdr_gui_node_libing0_18101_4696702751935649166 __log:=/home/libing/.ros/log/a227098c-fbc2-11e5-bd9c-28d244dc300f/stdr_gui_node_libing0_18101_4696702751935649166-5.log].
log file: /home/libing/.ros/log/a227098c-fbc2-11e5-bd9c-28d244dc300f/stdr_gui_node_libing0_18101_4696702751935649166-5*.log