turtlebot/turtlebot_simulator #54: [Kinetic] ROS nodes and Gazebo not linking

130s, 2016-06-17:
I might be doing something wrong, but let me just think out loud.

```
roslaunch turtlebot_gazebo turtlebot_world.launch
```

Nodes graph tells there's a breakage.
[image]

As a reference here's one I got on Indigo machine.

[image]

Obvisously due to this I can't e.g. send Twist command.

Some outputs.
```
$ rostopic list 
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/points
/camera/parameter_descriptions
/camera/parameter_updates
/camera/rgb/camera_info
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/clock
/cmd_vel_mux/active
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/switch
/cmd_vel_mux/input/teleop
/cmd_vel_mux/parameter_descriptions
/cmd_vel_mux/parameter_updates
/depthimage_to_laserscan/parameter_descriptions
/depthimage_to_laserscan/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/joint_states
/laserscan_nodelet_manager/bond
/mobile_base/commands/velocity
/mobile_base/sensors/bumper_pointcloud
/mobile_base/sensors/core
/mobile_base_nodelet_manager/bond
/rosout
/rosout_agg
/scan
/tf
/tf_static

$ rosnode list
/bumper2pointcloud
/cmd_vel_mux
/depthimage_to_laserscan
/gazebo
/laserscan_nodelet_manager
/mobile_base_nodelet_manager
/robot_state_publisher
/rosout

$ rosversion turtlebot_gazebo
2.2.2
$ rosversion gazebo_ros
2.5.7
$ rosversion gazebo_ros_control
2.5.7
```

For installation many packages are build from source (see https://github.com/turtlebot/turtlebot_apps/issues/143#issuecomment-226594335 my footage for that).

```
$ roslaunch turtlebot_gazebo turtlebot_world.launch 
... logging to /home/souflet/.ros/log/760b3eb8-34c5-11e6-97d3-0090f5bd194f/roslaunch-130s-serval-15387.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /home/souflet/link/ROS/kinetic_xenial/cws_household/src/turtlebot/turtlebot/turtlebot_description/robots/kobuki_hexagons_asus_xtion_pro.urdf.xacro

xacro.py is deprecated; please use xacro instead
started roslaunch server http://130s-serval:44680/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/souflet/link/...
 * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.2
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    depthimage_to_laserscan (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    laserscan_nodelet_manager (nodelet/nodelet)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [15406]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 760b3eb8-34c5-11e6-97d3-0090f5bd194f
process[rosout-1]: started with pid [15419]
started core service [/rosout]
process[gazebo-2]: started with pid [15443]
process[gazebo_gui-3]: started with pid [15450]
process[spawn_turtlebot_model-4]: started with pid [15457]
process[mobile_base_nodelet_manager-5]: started with pid [15458]
process[cmd_vel_mux-6]: started with pid [15459]
process[bumper2pointcloud-7]: started with pid [15460]
process[robot_state_publisher-8]: started with pid [15461]
process[laserscan_nodelet_manager-9]: started with pid [15491]
process[depthimage_to_laserscan-10]: started with pid [15506]
[ INFO] [1466193341.310227888]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1466193341.311206365]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1466193342.090442528, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1466193342.131507148, 0.070000000]: Physics dynamic reconfigure ready.
[ INFO] [1466193342.865785296, 0.250000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1466193342.872939698, 0.250000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[spawn_turtlebot_model-4] process has finished cleanly
log file: /home/souflet/.ros/log/760b3eb8-34c5-11e6-97d3-0090f5bd194f/spawn_turtlebot_model-4*.log
```

rohbotics, 2016-07-09:
Are you trying to simulate the create or the koubuki base?

130s, 2016-07-09:
Kobuki base for this.

rohbotics, 2016-07-09:
Add my fork of kobuki_desktop to your catkin_ws, and checkout the kinetic branch. This has all the modifications needed to get the gazebo plugin to compile and work under Gazebo7 (used with kinetic). 

https://github.com/rohbotics/kobuki_desktop/tree/kinetic
https://github.com/yujinrobot/kobuki_desktop/pull/45

130s, 2016-07-15:
I compiled https://github.com/rohbotics/kobuki_desktop/commit/e1e53436502f3fa0c54aaab6cd8620883d08a3ea in my catkin ws and tried the same command but still see the same issue. Thanks though.

> This has all the modifications needed to get the gazebo plugin to compile and work under Gazebo7 (used with kinetic).

I don't seem to have compilation issue. It seems to me a runtime issue.

NoviceMan, 2016-08-07:
sorry , my English is very poor. I have met the same problem with you ,did you have solved this problem now?  If you have，could you tell me the solution？Thank you very much.

stonier, 2016-08-10:
@rohbotics did you see the same runtime problem?

rohbotics, 2016-08-11:
I saw this, but it went way for me with the changes I made.

stonier, 2016-08-13:
Kobuki Gazebo Plugins is in the deb pipeline. For those having trouble, could you grab the deb from the shadow repository and see if that works for you?

130s, 2016-08-15:
Thank you @stonier, I confirmed that the issue is gone using the newest `kobuki_gazebo_plugins` from deb.
(I don't understand why source-building @rohbotics' code didn't work earlier, but I'm happy now :) and won't dig into it.)

Closing the issue since I'm the original reporter. Others who still see this issue feel free to add comments so that I or maintainers can re-open.

```
roslinguini@130s-serval:~/kinetic_xenial/cws_household$ rospack find kobuki_gazebo_plugins 
/opt/ros/kinetic/share/kobuki_gazebo_plugins
roslinguini@130s-serval:~/kinetic_xenial/cws_household$ rosversion kobuki_gazebo_plugins  
0.5.0
roslinguini@130s-serval:~/kinetic_xenial/cws_household$ rospack find turtlebot_gazebo
/home/roslinguini/kinetic_xenial/cws_household/src/turtlebot/turtlebot_simulator/turtlebot_gazebo
roslinguini@130s-serval:~/kinetic_xenial/cws_household$ roscd turtlebot_gazebo && git log 
commit 5e56b235e4fb3c8a8277d388194c4e3346901bd3
Author: Jihoon Lee <jihoonlee.in@gmail.com>
Date:   Wed Sep 16 11:57:06 2015 +0900

    2.2.2
```

closed 2016-08-15 by 130s