turtlebot/turtlebot_simulator #55: Release into kinetic

130s, 2016-07-14:
I'm sure it's in someone's todo list, and sorry if it's already WIP, but I open a ticket just because I didn't see it on [buildfarm's status page](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=turtlebot).

stonier, 2016-07-16:
I've got to get kobuki desktop fixed and released.  But coming very soon.

cerebro-deux, 2016-08-16:
Thanks @stonier for working on this.  Looking forward to it!

spragunr, 2016-08-16:
Thanks @stonier for this and for all of your work supporting Turtlebot. Is there an ETA for releasing turtlebot_simulator for kinetic?  I'm teaching a course in the fall and need to decide whether the lab can be updated to kinetic.  This seems to be the last missing piece.

tfoote, 2016-08-16:
https://github.com/ros/rosdistro/pull/12387

closed 2016-08-16 by tfoote