turtlebot/turtlebot_simulator #56: Simulation crash, TURTLEBOT_GAZEBO_WORLD_FILE undefined

bmagyar, 2016-08-18:
Hey guys,
The (wiki)[http://wiki.ros.org/turtlebot_gazebo/Tutorials/indigo/Gazebo%20Bringup%20Guide#Execution] instruction for starting up the gazebo simulation seems to be broken. When I hardcoded a value to turtlebot_gazebo into the launch file in question, things started working fine.

Wouldn't a solution similar to (this)[https://github.com/pal-robotics/tiago_simulation/blob/indigo-devel/tiago_gazebo/launch/tiago_gazebo.launch#L4] be safer with releases?

Btw I'm doing this on Kinetic.

stonier, 2016-08-18:
The environment variable `TURTLEBOT_GAZEBO_WORLD_FILE` is defined in the environment hooks. The only way you'll get an undefined variable is if that package hasn't been compiled and the environment hooks not installed OR you've compiled it, but haven't re-sourced your environment to get the environment variable itself. It *might* be a problem with Xenial/Kinetic's handling of environment hooks - I haven't tried it myself yet, but I doubt there's been change in that area.

Turtlebot follows this principle pretty much everywhere. It makes it far easier to know what is currently configured and what is configurable by simply running:

```
env | grep TURTLEBOT
```

The method you point out here is the earlier approach, which has the disadvantage in sometimes not being able to easily get in and configure the args, as well as having the important args scattered here and there, thereby making it hard to debug.

bmagyar, 2016-08-18:
Ok. Thanks for the answer. Indeed it started working just ok once I re-sourced after installing the debian packages.

closed 2016-08-18 by bmagyar