turtlebot/turtlebot_simulator #62: New release?

rohbotics, 2017-02-24:
The current version of turtlebot_stage available from debs is broken due to changes in the path to amcl. Looks like these bugs were already fixed in the from source code, but is still broken in the release version. A re-release should fix this.

130s, 2017-09-15:
Same from me.
In the following error I just encountered, the new version of the upstream (`turtlebot_navigation`) is already available as binary but deb of `turtlebot_gazebo` is still old and does not reflect the latest change already made in the source.

```
$ roslaunch turtlebot_gazebo turtlebot_world.launch
:
while processing /opt/ros/indigo/share/turtlebot_navigation/launch/includes/gmapping.launch.xml:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/opt/ros/indigo/share/turtlebot_navigation/launch/includes/gmapping.launch.xml'
The traceback for the exception was written to the log file
$ rosversion turtlebot_navigation
2.3.7
$ rosversion turtlebot_gazebo
2.2.2
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.5 LTS
Release:        14.04
```

rohbotics, 2017-09-16:
@tfoote Are you planning on running a release?

tfoote, 2017-09-18:
https://github.com/ros/rosdistro/pull/15939
https://github.com/ros/rosdistro/pull/15938

closed 2017-09-18 by tfoote