turtlebot/turtlebot_simulator #65: ‘LightPtr’ is not a member of ‘gazebo::physics’

wangzhaodong123, 2017-04-29:
When i catkin_make my workspace including turtlebot_simulator pkg, the problem happened:
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp: In member function ‘bool gazebo::GazeboRosApiPlugin::deleteLight(gazebo_msgs::DeleteLight::Request&, gazebo_msgs::DeleteLight::Response&)’:
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:783:3: error: ‘LightPtr’ is not a member of ‘gazebo::physics’
   gazebo::physics::LightPtr phy_light = world_->Light(req.light_name);
   ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:783:3: note: suggested alternatives:
In file included from /usr/include/gazebo-2.2/gazebo/msgs/MessageTypes.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/msgs/msgs.hh:24,
                 from /usr/include/gazebo-2.2/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-2.2/gazebo/physics/Model.hh:33,
                 from /usr/include/gazebo-2.2/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-2.2/gazebo/physics/physics.hh:2,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/msgs/light.pb.h:746:48: note:   ‘gazebo::msgs::LightPtr’
 typedef boost::shared_ptr<gazebo::msgs::Light> LightPtr;
                                                ^
In file included from /usr/include/gazebo-2.2/gazebo/common/Plugin.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/common/common.hh:28,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:35,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/rendering/RenderTypes.hh:70:38: note:   ‘gazebo::rendering::LightPtr’
     typedef boost::shared_ptr<Light> LightPtr;
                                      ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:783:29: error: expected ‘;’ before ‘phy_light’
   gazebo::physics::LightPtr phy_light = world_->Light(req.light_name);
                             ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:785:7: error: ‘phy_light’ was not declared in this scope
   if (phy_light == NULL)
       ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:799:27: error: ‘class gazebo::physics::World’ has no member named ‘Light’
       phy_light = world_->Light(req.light_name);
                           ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp: In member function ‘bool gazebo::GazeboRosApiPlugin::getLightProperties(gazebo_msgs::GetLightProperties::Request&, gazebo_msgs::GetLightProperties::Response&)’:
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1131:3: error: ‘LightPtr’ is not a member of ‘gazebo::physics’
   gazebo::physics::LightPtr phy_light = world_->Light(req.light_name);
   ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1131:3: note: suggested alternatives:
In file included from /usr/include/gazebo-2.2/gazebo/msgs/MessageTypes.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/msgs/msgs.hh:24,
                 from /usr/include/gazebo-2.2/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo-2.2/gazebo/physics/Model.hh:33,
                 from /usr/include/gazebo-2.2/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-2.2/gazebo/physics/physics.hh:2,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/msgs/light.pb.h:746:48: note:   ‘gazebo::msgs::LightPtr’
 typedef boost::shared_ptr<gazebo::msgs::Light> LightPtr;
                                                ^
In file included from /usr/include/gazebo-2.2/gazebo/common/Plugin.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/common/common.hh:28,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:35,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/rendering/RenderTypes.hh:70:38: note:   ‘gazebo::rendering::LightPtr’
     typedef boost::shared_ptr<Light> LightPtr;
                                      ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1131:29: error: expected ‘;’ before ‘phy_light’
   gazebo::physics::LightPtr phy_light = world_->Light(req.light_name);
                             ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1133:7: error: ‘phy_light’ was not declared in this scope
   if (phy_light == NULL)
       ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp: In member function ‘bool gazebo::GazeboRosApiPlugin::setLightProperties(gazebo_msgs::SetLightProperties::Request&, gazebo_msgs::SetLightProperties::Response&)’:
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1161:3: error: ‘LightPtr’ is not a member of ‘gazebo::physics’
   gazebo::physics::LightPtr phy_light = world_->Light(req.light_name);
   ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:1161:3: note: suggested alternatives:
In file included from /usr/include/gazebo-2.2/gazebo/msgs/MessageTypes.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/msgs/msgs.hh:24,
                 from /usr/include/gazebo-2.2/gazebo/physics/Entity.hh:28,
[... 31 lines cut ...]
                 from /usr/include/gazebo-2.2/gazebo/physics/physics.hh:2,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/msgs/light.pb.h:746:48: note:   ‘gazebo::msgs::LightPtr’
 typedef boost::shared_ptr<gazebo::msgs::Light> LightPtr;
                                                ^
In file included from /usr/include/gazebo-2.2/gazebo/common/Plugin.hh:43:0,
                 from /usr/include/gazebo-2.2/gazebo/common/common.hh:28,
                 from /home/wangzhaodong/ros/src/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:35,
                 from /home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
/usr/include/gazebo-2.2/gazebo/rendering/RenderTypes.hh:70:38: note:   ‘gazebo::rendering::LightPtr’
     typedef boost::shared_ptr<Light> LightPtr;
                                      ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:2407:29: error: expected ‘;’ before ‘light’
   gazebo::physics::LightPtr light = world_->Light(model_name);
                             ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:2408:19: error: ‘light’ was not declared in this scope
   if ((isLight && light != NULL) || (model != NULL))
                   ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:2422:68: error: ‘class sdf::SDF’ has no member named ‘Root’
     gazebo::msgs::Light msg = gazebo::msgs::LightFromSDF(sdf_light.Root()->GetElement("light"));
                                                                    ^
/home/wangzhaodong/ros/src/gazebo_ros/src/gazebo_ros_api_plugin.cpp:2449:31: error: ‘class gazebo::physics::World’ has no member named ‘Light’
       if ((isLight && world_->Light(model_name) != NULL)
Thanks for your attention,and i hope you can help me solve this problem.

tfoote, 2017-05-01:
Can you please clarify what versions of packages you are running? Such as Ubuntu, ROS, gazebo, turtlebot_simulator, and any other package in your installation or workspace tree that might be relevant to reproduce the error.

wangzhaodong123, 2017-05-06:
Thanks for advise.After installing gazebo7, I met another problem when i catkin_make my work_space:
Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake
Could you tell me why? Thank you.

closed 2017-05-06 by wangzhaodong123