turtlebot/turtlebot_simulator #67: In ros kinetic and gazebo 8, the gazebo is not subscribing the mobile_base/commands/velocity

onlytailei, 2017-05-23:
I installed gazebo8 and ros kinetic as the instructions from http://gazebosim.org/tutorials/?tut=ros_wrapper_versions#Gazebobuiltfromsource.

When I start the launch file in turltebot_gazebo.
It seemed that mobile_base/commands/velocity is not subscribed by gazebo node.
It never happens in gazebo 7/2 with ros indigo.
Then how can I control the robot in gazebo through ROS?

onlytailei, 2017-05-23:
The turlebot_simulator is compiled in workspace from source.

tfoote, 2017-05-26:
Can you please provide more specific details about how to reproduce your problem. What your workspace is and exactly the console outputs from your debugging that the node is not subscribing.

onlytailei, 2017-05-26:
Sorry, just find that I need to compile the kobuki also from the source as well.

hassan82, 2017-06-29:
Hi,
I could not compile the kobuki from source because it gave a lot of errors since some functions are deprecated. That's why I couldn't move the turtlebot using turtlebot_teleop package. Have you accomplished to move turtlebot?

onlytailei, 2017-06-29:
@hassan82  I just compile it from source. Nothing is wrong.
Remember to compile all of the dependencies from source.

hassan82, 2017-07-01:
Thanks @onlytailei for the help. I explain more about here to anyone who has this problem:
 
Install the **kobuki**, **kobuki_core**,  **kobuki_desktop**,  **kobuki_msgs** packages as _kinetic devel_ into your catkin workspace from:

[https://github.com/yujinrobot?language=&page=3&q=&type=&utf8=%E2%9C%93](url)

After that, run this command:

`rosdep install --from-paths src -i -y`

Finally, install **turtlebot** and **turtlebot_simulator** packages from:

https://github.com/turtlebot

Then `catkin_make`

tfoote, 2017-09-18:
Without more details such as debugging outputs, there's not enough information here to debug. I'm going to close this. If there's more details please comment back here and we can reopen.

closed 2017-09-18 by tfoote