turtlebot/turtlebot_simulator #70: Install on melodic

jinfagang, 2018-11-22:
Does this robot can upgrade to melodic?  As the Ubuntu 19.04 will come out, ubuntu18.04 should be support now

tfoote, 2018-11-26:
The TurtleBot packages have not been tested or released on melodic yet. I do expect the to be released there, but do not have a timeline when to expect it to land.

AnuragSahu, 2020-01-28:
The TurtleBot packages for melodic have been released and I am interested to contribute the melodic versions for turtlebot_simulator.

tfoote, 2020-01-30:
If you can test from source and validate that everything appears to be working we can make a release.

AnuragSahu, 2020-02-27:
The Release has been Made! The Melodic Branch is now available.

corot, 2020-02-27:
Sorry, I'm a bit confused.. what do you mean with the release has been made? I cannot see any turtlebot_xxxx pkg on http://repositories.ros.org/status_page/ros_melodic_default.html...
Nor any PR on https://github.com/ros/rosdistro

busFred, 2020-03-15:
I git clone the repository and set the environment variable. However, it seems like lacking a crucial package that is not compatible with ros melodic
```
fred@Fred-T420s:~/catkin_ws$ roslaunch turtlebot_stage turtlebot_in_stage.launch 
... logging to /home/fred/.ros/log/7ba48428-6668-11ea-8086-100ba9733a74/roslaunch-Fred-T420s-22880.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: turtlebot_bringup
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/fred/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file
```

teshanshanuka, 2020-06-12:
> I git clone the repository and set the environment variable. However, it seems like lacking a crucial package that is not compatible with ros melodic

Getting a similar issue. This searches for `turtlebot-xxxx` packages but melodic pkgs are `turtlebot3-xxxx`

jvaishnav, 2020-07-24:
I am facing the same issue as busFred and teshanshanuka with melodic distro. Has anyone found a solution?

corot, 2020-07-27:
You need to create a catkin workspace containing all the code listed on https://github.com/turtlebot/turtlebot_simulator/blob/melodic/turtlebot_simulator.rosinstall (plus dependencies).
[This wiki](http://wiki.ros.org/wstool) explains how to

jvaishnav, 2020-07-27:
Hello Jorge,
Thanks for getting in touch. Although i figured that the problem was for
melodic *"turtlebot3"  *is the package that i have to use in order to setup
the turtlebot and once i did that, it worked like a charm. Thank you. I
really appreciate it.
Kind regards,
Jay Vaishnav