turtlebot/turtlebot_simulator #74: Turtlebot sim not moving: [map] does not exist

DanielArnett, 2019-01-26:
I'm trying to drive the turtlebot in Gazebo. I've done this quite a few times, but now isn't working for some reason. I've done it before on Indigo, but struggling on Kinetic.

My installation procedure:

1. Install ROS Kinetic on Ubuntu 16.04
2. Install Catkin
3. Install gazebo

```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update
sudo apt-get install gazebo7
sudo apt-get install libgazebo7-*
```

Now I try to run gazebo

1. Run turtlebot gazebo
`roslaunch turtlebot_gazebo turtlebot_world.launch`
2. Run teleop
`roslaunch turtlebot_teleop keyboard_teleop.launch`

Now I send teleop commands but see no movement. So I check the topics and see if there is any movement:

1. `/turtlebot_node/cmd_vel` is responding to my teleop commands, the x velocity smoothly accelerates from 0.0 to 0.2 as expected.
2. `/cmd_vel_mux/input/teleop` does the same. With both of these I believe that turtlebot should now be moving. 

Here's my rqt_graph
[image]

Here's my frames
[image]

Notice that there is no /map frame.

When I open rviz I see
[image]

and this continues for every frame. So I add a static transform to try to fix this.
`rosrun tf static_tf_publisher 0 0 0 0 0 0 map base_footprint 100`

Now that error goes away but there's weirdness. The wheels still don't have the correct transformation?
[image]

So that's essentially where I am. The tfs aren't working by default. I've tried uninstalling, reinstalling, using different installation instructions on fresh partitions but I keep running into this unmoving turtlebot. Any help would be appreciated.

tfoote, 2019-01-29:
If you don't have a localization algorithm running the connectivity to the `map` frame is not expected. If you change the "Fixed Frame" in the rviz global options to something like `base_link` or `base_footprint`

[image]

DanielArnett, 2019-01-29:
That's fine but it still can't drive. I've tried teleop and some of the
standard autonomy packages with no luck. Again I had no issues on Indigo,
not that I think Kinetic is necessarily the problem.

tfoote, 2019-01-29:
I tried to reproduce this but it seems to teleoperate fine. Here's what I did. https://gist.github.com/tfoote/432813cd6f654b7ac6504a65fd5f27c4 

From this clean install there's likely things different about your environment that you will need to determine.

DanielArnett, 2019-01-29:
@tfoote Fantastic, thank you! I'll purge everything ROS related and give this a go.